pulp-platform/bender #4: Bender vsim fails with parallel builds

msfschaffner, 2019-04-03:
We may have to add a mechanism that prevents this failure case:

```
Another git process seems to be running in this repository, e.g.
an editor opened by 'git commit'. Please make sure all processes
are terminated then try again. If it still fails, a git process
may have crashed in this repository earlier:
remove the file manually to continue.
```

To add a little more context: I have a makefile-based simulation flow to rapidly explore different  configurations, and each simulation run is built using a `bender vsim` target. As long as I sequentialize the builds, everything works fine, but with parallel builds the git subprocesses clash.

fabianschuiki, 2019-04-04:
Good catch. Release 0.12 will add the ability to generate vsim compile scripts via the `bender script vsim` command (6e86aedb9a6723db1c7951c10367efa1900e5996). This will allow you to generate the script once, then run all your parallel simulations without invoking bender every time.

You can get the bleeding edge version of bender which already has the feature via `cargo install --git https://github.com/fabianschuiki/bender.git -f`.

fabianschuiki, 2019-10-16:
This is now solved by using the `bender script vsim` command.

closed 2019-10-16 by fabianschuiki