pulp-platform/bender #11: bender fails to install on Ubuntu 18.04

noureddine-as, 2019-07-26:
Hello,

Trying to install bender on Ubuntu 18.04, either from the Rust repos or from source, fails with the following error. Is Ubuntu 18.04 supported? (and Vivado 2018.1 ?)

Thanks in advance.

Best regards,

```bash
....
...
   Compiling serde_derive v1.0.97
   Compiling tokio-signal v0.1.5
   Compiling tokio-process v0.1.6
   Compiling bender v0.13.1 (/home/aitsaidn/PhD/playground/axspike/bender/bender-0.13.1)
warning: the item `canonicalize` is imported redundantly
   --> src/cli.rs:264:19
    |
7   | use std::fs::{canonicalize, metadata};
    |               ------------ the item `canonicalize` is already imported here
...
264 |     use std::fs::{canonicalize, metadata};
    |                   ^^^^^^^^^^^^
    |
    = note: #[warn(unused_imports)] on by default

warning: the item `metadata` is imported redundantly
   --> src/cli.rs:264:33
    |
7   | use std::fs::{canonicalize, metadata};
    |                             -------- the item `metadata` is already imported here
...
264 |     use std::fs::{canonicalize, metadata};
    |                                 ^^^^^^^^

warning: the item `fmt` is imported redundantly
   --> src/config.rs:493:13
    |
13  | use std::fmt;
    |     -------- the item `fmt` is already imported here
...
493 |         use std::fmt;
    |             ^^^^^^^^

warning: the item `Ordering` is imported redundantly
    --> src/sess.rs:1316:13
     |
28   | use error::*;
     |     -------- the item `Ordering` is already imported here
...
1316 |         use std::sync::atomic::Ordering;
     |             ^^^^^^^^^^^^^^^^^^^^^^^^^^^

warning: the item `Ordering` is imported redundantly
    --> src/sess.rs:1323:13
     |
28   | use error::*;
     |     -------- the item `Ordering` is already imported here
...
1323 |         use std::sync::atomic::Ordering;
     |             ^^^^^^^^^^^^^^^^^^^^^^^^^^^

error[E0283]: type annotations required: cannot resolve `std::string::String: std::convert::AsRef<_>`
   --> src/sess.rs:338:42
    |
338 |                     self.intern_string(k.as_ref()),
    |                                          ^^^^^^

error: aborting due to previous error

For more information about this error, try `rustc --explain E0283`.
error: failed to compile `bender v0.13.1 (/home/aitsaidn/PhD/playground/axspike/bender/bender-0.13.1)`, intermediate artifacts can be found at `/home/aitsaidn/PhD/playground/axspike/bender/bender-0.13.1/target`

Caused by:
  Could not compile `bender`.

To learn more, run the command again with --verbose.

```

accuminium, 2019-07-26:
Yes, Ubuntu 18.04 (as well as Vivado 2018.1) should work. The error you get is due to a regression in Rust 1.36.0, which we are aware of and will fix soon. For now, you can either build bender locally with Rust 1.35.0 (e.g., using rustup: `rustup install 1.35.0; rustup run 1.35.0 cargo build --release`) or download the pre-built (with Rust 1.35.0) generic Linux binary for Bender 0.13.1 from https://github.com/fabianschuiki/bender/releases/download/v0.13.1/bender-0.13.1-x86_64-linux-gnu.tar.gz.

noureddine-as, 2019-07-26:
Thanks @accuminium , this worked, and I succeed to launch `bender script vsim > script.tcl`

Concerning Vivado support, when I launch `bender script vivado > script.tcl` it says that it only supports vsim and synopsys, how can I generate a Vivado TCL script?

Is there any documentation on that (other than this [pdf](https://pulp-platform.org/docs/riscv_workshop_zurich/bender_wosh2019.pdf)) ? or examples ?

Thanks again!
Best regards,

accuminium, 2019-07-26:
You are right, `script vivado` is not implemented yet. I confused that with an internal script we have been using, sorry. We plan to implement `script vivado`, but due to Vivado's limitations on source sets, includes, and defines, this is unfortunately quite tricky.

In the meantime, I can offer you the following workaround: First, generate a text file that contains the path of one source file per line
```sh
bender sources -f | grep -Po '(?<=      ").*\.sv(?=)' > sources.txt
```
Then, in Vivado, read that file into a list and call `add_files` on that list:
```tcl
set f [open sources.txt]
add_files [split [string trim [read $f]]]
close $f
```
Note that this will not set the proper defines and include directories yet, which you have to do manually in a further step.

For examples, I suggest you look at `Bender.yml` files in our repositories, e.g., in [`common_cells`](https://github.com/pulp-platform/common_cells/blob/b93f917100056abd211b8548225cfe3b41a7094f/Bender.yml).

noureddine-as, 2019-07-26:
Okay, thank you very much for the trick and the additional informations @accuminium !

Best regards,

closed 2019-07-26 by noureddine-as

accuminium, 2019-07-30:
To follow up on this, the compilation regression with Rust 1.36.0 has been fixed in Bender 0.13.3, and `bender script vivado`, including global include directories and defines (#14) is now implemented on `master` (and is expected to be part of 0.14 soon).