pulp-platform/bender #23: Vivado script optional no Synthesis

WRoenninger, 2020-03-05:
`bender script vivado` adds the `TARGET_SYNTHESIS` regardless of test flags.

fabianschuiki, 2020-03-05:
Do you think @WRoenninger it is a use case to ever use Vivado for purposes other than synthesis?  What's your take on this @accuminium ?

accuminium, 2020-03-10:
@fabianschuiki Yes, Vivado is also used to simulate designs with Xilinx libraries, either in the embedded simulator or as "proxy" to provide a third-party simulator with the required libraries.  I think we should support this.

Should we add an extra format (e.g., `vivado-sim`) that invokes the existing Vivado TCL emitter (`emit_vivado_tcl`) with different `format_targets` (i.e., `simulation` instead of `synthesis`), or should we add a flag (e.g., `--simulation`) specific to the `vivado` format?

fabianschuiki, 2020-03-10:
Yeah something like this would be pretty nice!

PR #26 (2020-03-11): script: Add `vivado-sim` format
Closes #23.

### Added
- Add `vivado-sim` format to the script command.  This format differs from the `vivado` format in that it targets simulation, not synthesis; other than that, it is equivalent to the `vivado` format.
closed 2020-03-11 by fabianschuiki