pulp-platform/bender #25: Dependency Patch version could be optional

WRoenninger, 2020-03-10:
When having the dependency:
```
riscv-dbg:          { git: "https://github.com/pulp-platform/riscv-dbg.git",          version: 0.3                   }
```
Bender complains:
```
error: Dependency `riscv-dbg` from `https://github.com/pulp-platform/riscv-dbg.git` cannot satisfy requirement `^0.3`
```

accuminium, 2020-03-11:
`0.3` [is not a Semantic Version number according to the definition](https://semver.org/#spec-item-2).  That is why the `semver` package correctly does not parse it at
https://github.com/fabianschuiki/bender/blob/cd41a2a44517e22769dca9a5950db2d8359c7d9c/src/sess.rs#L555

@fabianschuiki: What is your view on this? Should we force package owners to define proper Semantic Version numbers, or should we assume `0.3` is equivalent to `0.3.0` (but what do we do if someone defines *both* `0.3` and `0.3.0`?) and extend accordingly before invoking the `semver` parser?

fabianschuiki, 2020-03-11:
That's a good catch. @accuminium I think your solution is nice: just assume `0.3` is equivalent to `0.3.0`, and then maybe if both are defined, opt for `0.3.0` as the "correct" one.

accuminium, 2020-03-13:
The central disadvantage of the "assuming `v0.3` == `v0.3.0` unless there is a `v0.3.0`" solution to me is that reproducibility is lost if a repository has `v0.3` but not `v0.3.0` when first users check it out, then later `v0.3.0` to a different commit than `v0.3` when other users check it out. Bender would report identical versions while the actual commits would be different.

I think we should not opt for that but instead ask repository owners to create SemVer-compliant tags.  They can still keep their own-style tags if they want, and creating a few extra tags is little effort.

fabianschuiki, 2020-04-27:
@accuminium You raise a valid point regarding the potential impact of simply assuming `0.3` to be `0.3.0`. Let's rather leave this with the repository owners. It's not hard to update a tag after the fact, so it's not like we are preventing any use by requiring proper tags.

closed 2020-04-27 by fabianschuiki