pulp-platform/bender #45: Add version requirement information to bender parents command

meggiman, 2021-01-27:
For recursive bumping of dependencies, the `parents` command is already really useful. However, a flag I am really missing for this command is the option to print the (semantic) version requirements along with the parents. This would make it much easier to identify parent packages that need version bumping.

accuminium, 2021-01-27:
/cc @micprog

accuminium, 2021-01-27:
Thanks for this feedback, @meggiman !

micprog, 2021-01-27:
I started implementing this, however was unsure how to appropriately fetch the version requirement. Completing the following TODO
https://github.com/fabianschuiki/bender/blob/c02bb93806fdfe8d0e030cbe0592e01d3ce354dc/src/cmd/parents.rs#L38
and adding appropriate prints to the following command
https://github.com/fabianschuiki/bender/blob/c02bb93806fdfe8d0e030cbe0592e01d3ce354dc/src/cmd/parents.rs#L49-L51
should fix this issue.

PR #47 (2021-02-02): Update parents command
Add version requirement print to `parents` command to fix #45, minor formatting changes
commit 71c6a9c9 (2021-02-02): Adds version print to `parents` command, fixes #45
closed 2021-02-02 by fabianschuiki