pulp-platform/bender #60: Bender panicks when Bender.lock is outdated

gigantenbein, 2021-12-06:
When adding a new package to `Bender.yml` and a `Bender.lock` file has already been generated, Bender fails and exposes a 
```
thread 'main' panicked at 'no entry found for key', src/sess.rs:181:73
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
It can be solved by either removing `Bender.lock` or running `bender update`. A direct indication for these steps in the above error message would be nice.

To reproduce the error, follow these steps for a minimal example
1. `mkdir bender_test` and navigate to this directory.
2. Create a `Bender.yml` in `bender_test`
```
package:
  name: bender_test
dependencies:
  common_cells: { git: "https://github.com/pulp-platform/common_cells.git",  version: 1.23.0 }
```
3. Run `bender script vsim`
4. Add a new package to your `Bender.yml`, e.g.
```
package:
  name: bender_test

dependencies:
  axi:          { git: "https://github.com/pulp-platform/axi.git",            version: 0.29.1 }
  common_cells: { git: "https://github.com/pulp-platform/common_cells.git",   version: 1.23.0 }
```
5. Run `bender script vsim` again, this should cause the panick.

micprog, 2021-12-06:
https://github.com/pulp-platform/bender/commit/7823d545b5113b7b48be1614bc7a13ce1ce8f1cf fixes the issue, giving an error to run `bender update` if the dependency names in Bender.yml are not found in the lock file.

micprog, 2021-12-09:
Fixed by #62

closed 2021-12-09 by micprog