pulp-platform/bender #78: [request] Dynamic Source File Lists

Suudy, 2022-02-25:
It appears the file lists in the Bender.yml are static.  Perhaps some support to call a hook (Python?  Shell Script?  Command?) that returns a list of source files.

The specific use case is for the Xilinx IP generators where one might invoke Vivado, create a core, then export the list of files.

Perhaps syntax such as:

sources:
  - @(<command to execute>)
  - $(<command to execute>)

accuminium, 2022-02-26:
Hi @Suudy, thanks for your suggestion!

I can see the use of constructing the source file list through commands. However, I think letting Bender execute an arbitrary command provided in a manifest file would be catastrophic from a security perspective, as it essentially forms a [Shellcode threat](https://en.wikipedia.org/wiki/Shellcode).

Could your use case instead be implemented by adding an *include* command to the `sources` section of the Bender manifest? Upon encountering such an *include* command with a path argument, Bender would read the file at the path and insert each line in that file as source path in the manifest, at the position of the *include* command. The file to be included could then be generated through any command the user desires, but the user has to execute the command themselves (or at least the command has to be part of a build flow that the user executes, and it is not provided by some dependency).

Suudy, 2022-02-26:
I’m not seeing a concern for shell code exploit. This is all executed within the context of a project which are under the control of developer.  Unless you are concerned with remote repositories owned/maintained by third parties?

in that case, perhaps a global enable that has to be turned on to support execution of code?

This feature is desirable within bender because it automatically determines which components need dynamic file lists without a manual step.  I’m not sure how a large project might navigate the generation of these include lists.

accuminium, 2022-02-26:
> Unless you are concerned with remote repositories owned/maintained by third parties?

Yes. You would have to trust the developers of any dependency (and of their dependencies, etc.) to run any command on your machine.

> This feature is desirable within bender because it automatically determines which components need dynamic file lists without a manual step. I’m not sure how a large project might navigate the generation of these include lists.

What makes the file list dynamic? Is RTL code generated from higher-level code? Can the RTL code files not be enumerated a priori and/or can they not be checked into a repository that forms a Bender package?

Suudy, 2022-02-26:
The example I can give is in Vivado.  What I typically do is write some TCL to generate a core and then use a TCL command to export the list of files.  That list of files can change depending on the version of the core, whether it is configured for simulation or synthesis, etc.  Normally I do a single TCL file for the whole project, then call individual TCL files for each generated core.  I then gather all the sources files, calling `get_files` to get the source files.

While they could be enumerated in advance, the list can change depending on the context or even on the version of the core.