pulp-platform/bender #91: Allow wildcards for sources

lhall-sol, 2022-05-12:
I imagine I'm not the only person who would be adding all files in a directory. Having to manually specify the path for each one seems unnecessary, when it could be done with something like:
```
sources:
 - src/*.vhd
 - src/**/*.sv
```

micprog, 2022-05-12:
I don't think wildcard references are always the best approach, as in some cases wrong files may be included. However, feel free to propose a change in a pull request.

lhall-sol, 2022-05-12:
It's certainly not always the best approach, but I think there's some clear use cases. I've never used rust, but I might have a go at implementing it eventually.

andreaskurth, 2022-08-03:
One problem with wildcards is that file order is implicit (usually alphanumerical). Some EDA tools need an ordered list of files in which any file can only use modules and packages from other files earlier in the list. So wildcards will not work for all EDA tools and repositories.

micprog, 2025-10-06:
Added in #203

closed 2025-10-06 by micprog