pulp-platform/bender #92: Allow relative git dependencies

lhall-sol, 2022-05-12:
Git submodules allow the use of a relative path (https://git-scm.com/docs/git-submodule) and is encouraged by GitLab (https://docs.gitlab.com/ee/ci/git_submodules.html). It has a big benefit of allowing the authentication method to be inferred.

It would be great if bender worked with those same relative paths. I believe it would need to get the remote of the current package's repository and resolve the path (removing any instances of /../) to get an absolute one, before continuing as normal.

micprog, 2022-05-13:
Bender is designed to work independently of git submodules, as a single package may require two different packages that in turn again require a single (fourth) package, albeit referencing different versions. If managed with submodules, package reuse and updates would become significantly more complicated, as submodules would then include multiple versions of the same dependency, or require the top-level to manage these submodules. Furthermore, if different versions are referenced, as far as I know submodules will reference a commit, while bender will parse semantic versioning tags and select a compatible version, or prompt you to select a version. As an example, `common_cells` is a widely used package, where each dependency may reference a different version of `common_cells`, and managing this solely with submodules is almost impossible, further complicated by the fact multiple identical packages may be included.

Regarding authentication, I suggest to use `https` links when referencing dependencies, as this allows checkout for all public repositories. Private repositories may need authentication anyways.

I hope this explanation makes sense, if I understood the issue wrong feel free to clarify and open a pull request with the corresponding changes.

lhall-sol, 2022-05-13:
I guess my suggestion wasn't clear. I'm not suggesting the use of submodules. I'm suggesting that the relative url functionality of submodules is really good and would be great to have the same behavior in Bender. That way, authentication isn't an issue for dependencies on the same server. 

For example if I had a private project in "gitlab.privatecompany.com.au/group/project.git" and added a git dependency with the URL "../dependency.git". That URL would then be evaluated to the correct absolute URL, it wouldn't matter if it was SSH or HTTP or even including a Personal Access Token. 

I believe it would only require an extra step for resolving the URL and everything else would be the same within Bender.