pulp-platform/bender #104: Cannot find the repo cloned with ``bender import``

znaky, 2022-10-05:
I am running bender import to clone a repo that doesn't support bender. Everything looks good. It really seems like cloning it to /tmp and then checking out the correct commit. But I cannot locate the cloned repo afterwards. My ``Bender.yml`` looks like below. Am I missing something?
```
# Package metadata. Required.
package:
  # The name of the package. Required.
  name: my_package

# List of imported files from external repositories not supporting bender. Optional
external_import:
    # package name
  - name: my_external_repo
    # target directory
    target_dir: external/my_external_repo
    # upstream dependency (i.e. git repository similar to dependencies)
    upstream: { git: "ssh://git@code.mycompany.com/fpga/my_external_repo.git", rev: "224ace1221ebe16fec104c58250faa9f4eac84a2" }
```

FYI, when I add mapping, it just cannot find the files and errors out.

micprog, 2022-10-05:
The `import` feature in bender requires a mapping to be defined to include parts of the upstream repository into your project. The repository is cloned into a temporary location, and the needed files are copied over afterwards, so it makes sense that you cannot find the repository.
Try adding the `--refetch` flag to the `bender import` command, as this will trigger the proper update of your files. Otherwise bender will attempt to print a diff between the files included in your repository and those in the upstream, and here it cannot find the needed directory.
If this fixes the issue, I will update the documentation, errors, and code to provide a bit better user experience for this feature.

closed 2022-10-05 by znaky

micprog, 2022-10-05:
You can link directories, so you could try and give `'.'` a shot, but I haven't tested that yet. It might make sense to by default clone the full repository if no mapping is provided, but that is not implemented yet.

znaky, 2022-10-05:
>  It might make sense to by default clone the full repository if no mapping is provided

I agree.

I tested ``'.'`` , ``folder_layer1`` and ``folder_layer1/folder_layer2/last_folder`` cases and only the last one is working. i.e. it cannot link recursively. Maybe, *linking recursively* can be another feature that may be added in the future releases. 

Thanks again

znaky, 2022-10-05:
Found it. If I do it like below, it works. **But,** it still throws an error and looks like skips the ``.txt`` files. Interesting.
```
    mapping: 
      - {from: '', to: '' }
```
```
error: Cannot startup runtime. No such file or directory (os error 2)
```