pulp-platform/bender #114: Bug: Bender sources may start with `false`

flaviens, 2023-01-23:
Hi!

## Brief

Since a recent version of Bender, a parasitic `false` appears at the beginning of `bender.sources`, making them uninterpretable by Morty. I tried at least with the current `0.27.0` version.

## How to reproduce

```
git clone https://github.com/openhwgroup/cva6.git
bender sources -f -t ariane > my.sources
head my.sources
```

And Morty (at least `0.8.0`) does not like it:
```
morty -f my.sources
error parsing json in `my.sources`: invalid type: boolean `false`, expected a sequence at line 1 column 5
```

## Suggestion

There was a similar issue with similar tools (sv2v I think). I'd suggest either or both:

* adding a `-o target_file` (or another letter) to avoid using stdout as the  way of writing to the output file.
* removing what seems to be a debug print.

## Notes

It does not seem to happen with all designs.

Thanks!
Flavien

micprog, 2023-01-24:
Does #115 fix your issue?

flaviens, 2023-01-24:
Looks very probable, thanks!

closed 2023-01-24 by flaviens