pulp-platform/bender #116: Rerunning the same bender command twice produces different results

epsilon537, 2023-01-27:
When I run 'bender script verilator' twice on my source tree, I get slightly different results. The files are reordered:

First time:
```
...
+define+TARGET_SYNTHESIS
+define+TARGET_VERILATOR
+incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_
prim_util_get_scramble_params_0/rtl
+incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_
ibex_ibex_core_0.1/rtl
+incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/vendor/lowrisc_ip/ip/prim/rtl
+incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/vendor/lowrisc_ip/dv/sv/dv_utils
+incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_
prim_util_memload_0/rtl
...

```
Second time:

```
+define+TARGET_SYNTHESIS
+define+TARGET_VERILATOR
+incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/vendor/lowrisc_ip/dv/sv/dv_utils
+incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_
dv_dv_fcov_macros_0
+incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_
prim_util_get_scramble_params_0/rtl
+incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_
prim_util_memload_0/rtl
+incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_
ibex_ibex_core_0.1/rtl
+incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_
prim_assert_0.1/rtl
+incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/vendor/lowrisc_ip/ip/prim/rtl
...
```

Any idea why this happens and if there's anything I can do to prevent it?

These are the relevant bender.yml manifests:

https://github.com/epsilon537/boxlambda/blob/develop/gw/projects/hello_world/Bender.yml
https://github.com/epsilon537/boxlambda/blob/develop/gw/components/ibex_wb_core/Bender.yml
https://github.com/epsilon537/boxlambda/blob/develop/gw/components/ibex/Bender.yml

I'm using bender 0.27.1

Thanks for any help.

micprog, 2023-01-27:
An important requirement for Bender is that the build order of files is preserved, however this is not as critical for include directories. Nonetheless, I understand your concern. Could you try out the `indexmap` branch to see if this fixes your issue? The changes replace the use of `Hash` structures with ordered `Hash` structures, however the issue may also be due to the async functions.

epsilon537, 2023-01-27:
Thanks for the quick response. Unfortunately, I see the same issue on the indexmap branch:

```
epsilon@LAPTOP-BQA82C62:~/work/bender$ git status
On branch indexmap
Your branch is up to date with 'origin/indexmap'.

nothing to commit, working tree clean
epsilon@LAPTOP-BQA82C62:~/work/bender$ which bender
/home/epsilon/.cargo/bin/bender
```

```
epsilon@LAPTOP-BQA82C62:~/work/boxlambda/gw/projects/hello_world$ bender script verilator > out1.txt
epsilon@LAPTOP-BQA82C62:~/work/boxlambda/gw/projects/hello_world$ bender script verilator > out2.txt
epsilon@LAPTOP-BQA82C62:~/work/boxlambda/gw/projects/hello_world$ diff out1.txt out2.txt
8a9
> +incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/vendor/lowrisc_ip/dv/sv/dv_utils
10d10
< +incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_dv_dv_fcov_macros_0
11a12
> +incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_prim_assert_0.1/rtl
12a14
> +incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_dv_dv_fcov_macros_0
14,15d15
< +incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/build/lowrisc_ibex_top_artya7_0.1/src/lowrisc_prim_assert_0.1/rtl
< +incdir+/home/epsilon/work/boxlambda/gw/components/ibex/../../../sub/ibex/vendor/lowrisc_ip/dv/sv/dv_utils
epsilon@LAPTOP-BQA82C62:~/work/boxlambda/gw/projects/hello_world$ which bender
/home/epsilon/.cargo/bin/bender
```

/home/epsilon/.cargo/bin/bender is where the indexmap version of bender is located.

micprog, 2023-01-27:
Very strange behavior. I attempted to trace the issue a bit further, with the newest commit on the indexmap branch I no longer see the issue you were reporting, but that might just be by chance...

epsilon537, 2023-01-27:
I pulled in your latest commit on the indexmap branch just now, but I still see the issue.
Should you want to reproduce the issue on my code base, you can do the following:

1. clone the repo: https://github.com/epsilon537/boxlambda/
2. Switch to the develop branch
3. cd gw/projects/hello_world
4. bender script verilator > out1.txt
5. bender script verilator > out2.txt

micprog, 2023-01-27:
I cloned and tested in your repository and (after also running `bender update` to update the `Bender.lock` file) I was able to confirm the issue in the previous version, but I no longer see the issue with the latest commit. 

When using the development version of bender, make sure to properly install it with cargo (either with `cargo install --path .` or `cargo install --git https://github.com/pulp-platform/bender.git --branch indexmap`), or run the development build using `cargo run -- -d PATH_TO_TOP_PACKAGE DESIRED_COMMAND` from bender directory. If you want, I just added a commit to identify the indexmap branch using the `bender --version` command to double-check the correct version is used.

epsilon537, 2023-01-27:
That helped! I'm not familiar with the rust toolchain. I probably made a pilot error when building with the commit that fixes the issue. I rebuilt and confirmed the version. Now there's no longer a diff when running 'bender script verilator' twice.
Thank you very much for your help, and thank you for making Bender available :-)

closed 2023-12-29 by micprog