pulp-platform/bender #127: How to create a package?

hydroid7, 2023-06-01:
Hi there,
while implementing siphash for my master thesis, I discovered the PULP platform. I'd like to create a `bender` package from it to allow other users to use my code.
Since there is no documentation on how to create a package, I'd want to kindly ask you on some help how to create one from my code?

[https://github.com/hydroid7/siphash](https://github.com/hydroid7/siphash)

Thank you in advance.
Best Regards

micprog, 2023-06-01:
In general, to create a package, all you will have to do is add a `Bender.yml` file to your repository with the format described [here](https://github.com/pulp-platform/bender#manifest-format-benderyml). There is no central repository or collection for bender packages; we also manage them using the git repositories in the PULP platform. 

Regarding your [`bender.yml`](https://github.com/hydroid7/siphash/blob/master/bender.yml) file, AFAIK, you should capitalize the `B` in the file name, the rest should work. Other than that, as you do not have any dependencies, I don't think declaring the `checkout_dir` is necessary (this indicates where bender should place dependencies if this is your top-level package). Feel free to check out more complex examples like [AXI](https://github.com/pulp-platform/axi/blob/master/Bender.yml).

hydroid7, 2023-06-01:
Thank you for the fast reply! :) Indeed renaming the file helped. The example is also really helpful.

closed 2023-07-07 by hydroid7