pulp-platform/bender #130: Bender vendor support for upstream versions

colluca, 2023-08-10:
Version 0.27.1 does not seem to support specifying `upstream` dependency revisions using the `version` attribute.

With the following `Bender.yml` contents:
```
vendor_package:
  - name: musl
    target_dir: sw/math
    upstream: { git: https://github.com/kraj/musl.git, version: 1.2.4 }
    include_from_upstream:
      - "src/math/tanh.c"
```
the `bender vendor init` command fails.

It succeeds when specifying the commit hash using the `revision` attribute:
```
vendor_package:
  - name: musl
    target_dir: sw/math
    upstream: { git: https://github.com/kraj/musl.git, rev: 7a43f6fea9081bdd53d8a11cef9e9fab0348c53d }
    include_from_upstream:
      - "src/math/tanh.c"
```

I found this comment to be a bit misleading:
https://github.com/pulp-platform/bender/blob/dc733b2c63f535efc24a08ee35adbc308f1946de/README.md?plain=1#L185

micprog, 2023-08-10:
Bender intentionally does not support `version` tags for `vendor`ed dependencies. To enforce a consistent upstream, a commit hash must be used. lowRISC's `vendor.py` script enforces a consistent version by adding an additional `.lock` file, however, for bender's `vendor` feature we opted to enforce a commit hash. See the error message in https://github.com/pulp-platform/bender/blob/dc733b2c63f535efc24a08ee35adbc308f1946de/src/cmd/vendor.rs#L118

micprog, 2023-09-05:
Fixed in #131 by https://github.com/pulp-platform/bender/pull/131/files#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5R185

closed 2023-09-05 by micprog