pulp-platform/bender #140: Possibility to release a build for macOS?

V0XNIHILI, 2023-10-19:
Dear Bender team,

I am really excited by the possibility of a well-built and widely adopted Verilog package manager. So, thank you very much for developing this project!

I was wondering whether it would also be possible to release pre-built binaries for macOS (the platform that I am using).

Best,
Douwe

micprog, 2023-10-23:
Hi Douwe,

Thank you for your interest in bender! I'd be happy to add a build flow for macOS, but currently do not have the time to implement it properly. Ideally, a universal binary (both for `x86_64` and `arm64`/`aarch64`) would be implemented and the download script (found on the `website` branch) adjusted to run correctly for macOS. I would be happy to review a PR that adds this feature! Otherwise, you can always [install rust](https://www.rust-lang.org/tools/install) and run `cargo install bender` to have a system installation for bender.

Best,
Michael

V0XNIHILI, 2023-10-24:
Hey, for now I indeed used `cargo install bender` and this was really easy and works perfectly! I don't have enough knowledge to setup a build flow for macOS unfortunately :(

micprog, 2025-10-06:
Added in #145 and #146

closed 2025-10-06 by micprog