pulp-platform/bender #142: `plugins:` will be deprecated?

mac-edmondson, 2023-12-08:
Just now seeing this in the readme.
```yaml
# Format is the same as `dependencies` in a package manifest.
# DEPRECATED: This will be removed at some point.
plugins:
  additional-tools: { path: "/usr/local/additional-tools" }
```

Wondering if this is true and what the reasoning behind this is? I have found this feature quite useful. I have made packages which act as a means to run tooling via bender. For example, all of my components require a unit testing dependency that allows me to simply kick off a unit test, for any of my components, with something along the lines of `bender rtl_sim <args>`.  Anyway, I have found the feature quite useful and would be sad to see it go :(.

mac-edmondson, 2023-12-09:
Looking back at this tonight, is this only for the configuration files that this feature will be removed? Or does this include the manifest file? I use this feature in my manifests. Thanks.

micprog, 2023-12-11:
Hi @mac-edmondson,
Currently, there are no plans to remove this feature.
For some reason on the comment, `plugins` were originally introduced as a quick way to get scripts prior to `script`s being introduced as their own command. Afterward, we had no specific use for plugins, but left them within `bender` in case they were useful to others, albeit without continually testing their functionality.
I hope `plugins` continue to be useful for you, feel free to contribute fixes if they do break.

closed 2023-12-11 by micprog