pulp-platform/bender #144: using environment variables in path dependencies

rbonthond, 2023-12-11:
Would it be possible to use environment variables instead of relatives paths in Bender.yml for package path dependencies? We're trying to re-use some things between different projects with different layouts, and being able to use environment variables would make this much more flexible.

```
package:
    name: A

dependencies:
   B: {path: "$WORKSPACE/B"}

sources:
  - files:
    - a.txt

```

micprog, 2023-12-29:
Using environment variables for a path may cause issues, especially when sharing a project across devices where the environment variables may not be set up. Nonetheless, does #147 enable this feature for you? I haven't tested it yet, but internally bender will work continue working with full paths, evaluating the path only when initially parsing the Bender.yml (for dependencies only on update).

rbonthond, 2024-01-09:
thank you very much #147  works great for me!

closed 2024-01-09 by rbonthond