pulp-platform/bender #159: Bender ignores version tags

otoomey, 2024-03-13:
Bender ignores version tags in Bender.yml. For instance, the following 
```yml
dependencies:
  common_cells:       { git: https://github.com/pulp-platform/common_cells,       version:  1.28.0  }
```
does not use release 1.28.0, defaulting to the latest release instead.

otoomey, 2024-03-13:
I believe I've narrowed this down a bit further. I have another dependency that requires `common_cells` version ^1.31.0, and Bender seems to be overruling my dependency declaration and just grabs the newest release.

micprog, 2024-03-13:
Bender always chooses the most recent, compatible version, which is especially useful when specifying multiple versions, attempting to align the different dependency requests. To limit the version used, you can use other specifiers, such as `=`, `~`, or `<`. Internally, rust's [semver](https://docs.rs/semver/latest/semver/) package is used to parse the version requirements, like to [cargo dependencies](https://doc.rust-lang.org/cargo/reference/specifying-dependencies.html). As an example, you can set the version as follows:
- `version: "1.28.0"`: limits to 1.x.y, where x >= 28, choosing the newest available
- `version: "~1.28.0"`: limits to 1.28.x, where x is the newest available
- `version: ">=1.28.0, <1.31.0"`: limits as specified
This is all already explained in the [README](https://github.com/pulp-platform/bender?tab=readme-ov-file#dependencies)

closed 2024-03-13 by otoomey