pulp-platform/bender #161: Package with only include_dirs doesn't work

msmithTT, 2024-04-10:
I have a package that just contains include files (global files that define parts of a larger project).  If I attempt to create a Bender file for this, for example:

```
package:
  name: my_package
  authors:
    - "My Name"

sources:
  include_dirs:
    - rtl

  files:
```

This doesn't generate any +incdir+ command-line options.

If I add a single dummy file to this, then it adds the required +incdir+ command-line options as well as the dummy file:

```
package:
  name: my_package
  authors:
    - "My Name"

sources:
  include_dirs:
    - rtl

  files:
    - rtl/empty_dummy_module.sv
```

micprog, 2024-04-10:
I think you may be looking for the `export_include_dirs` functionality.

msmithTT, 2024-04-11:
Thanks for the hint, but unfortunately doesn't make a difference.

Using this Bender file:

```
sources:
  export_include_dirs:
    - rtl

  files:
```

Then running 'bender script flist' it returns an empty string.

micprog, 2024-04-11:
You may want to try the `flist-plus` script format, which also adds plusargs: `+incdir+` and `+define+`. The `flist` script type only returns files.

commit 34a6b248 (2024-04-17): Keep properties for sourcegroups without files

Fixes #161