pulp-platform/bender #169: Tabs vs spaces

kasun-buddhi, 2024-04-26:
When I was using Bender, I was confused with using indentations. I think Bender only uses spaces as indentation. But I think placing white space (tabs or spaces) is much easier for programmers.

micprog, 2024-04-26:
I assume you mean in the generated scripts? I don't quite know how you would like to tailor to users' preferences, but feel free to propose a change in a PR.

kasun-buddhi, 2024-05-16:
Not in the generated scripts. When I was making a bender.yml file, I faced that issue. It is not a major issue but It would be great to have white spaces instead of tabs and spaces.

micprog, 2024-05-17:
The `Bender.yml` is read using the [`serde_yaml`](https://docs.rs/serde_yaml/latest/serde_yaml/) package (see https://github.com/pulp-platform/bender/blob/master/src/cli.rs#L375-L387). If there is a configuration that is missing to allow your whitespaces, let me know (easiest with a PR that can be integrated), but fixing `serde_yaml` within bender does not make sense.