pulp-platform/bender #170: overrides doesn't seem to work in specifying a different branch - what am I doing wrong?

msmithTT, 2024-05-06:
I have a CPU that I'm putting into Bender, and it relies upon a NOC module.  So the CPU's Bender file includes a dependency:

```
dependencies:
  tt_noc:             {git: "git@yyz-tensix-gitlab:tensix-hw/hw-noc.git", rev: "foobar"}
```

In this case, foobar is the name of the branch that I want it to pull the RTL from for the tt_noc.

However, in an upper level, I want to pull the RTL for tt_noc from a different branch to test some newer code.  So I specify an override:

```
overrides:
  tt_noc:    {git: "git@yyz-tensix-gitlab:tensix-hw/hw-noc.git",                                rev: foobar_new_stuff}
```

At the upper level, I want to use the CPU, but instead have it pull the tt_noc code from the foobar_new_stuff branch rather than the default foobar branch.  But, this override doesn't seem to do anything.  What am I doing wrong?

micprog, 2024-05-07:
Specifying overrides is only possible at the top package in a `Bender.local` file (config file, not manifest, see in [ReadMe](https://github.com/pulp-platform/bender?tab=readme-ov-file#configuration-format-benderyml-benderlocal)). Afterwards, you will need to re-run `bender update` to ensure the dependencies are appropriately updated. If multiple sources are specifying a dependency, bender will find a compatible version, use the override if present, or ask for user input if no compatible version can be found (i.e., for non-identical branches). After updating, you can also check these requirements with a `bender parents tt_noc` command, where package requirements and overrides should be shown.

As a side note, we generally use commit hashes and tag names (or version tags) for dependencies, not branch names, as a branch can be updated with incompatible changes while hashes can not and tags should not have updates.

closed 2024-05-07 by msmithTT

msmithTT, 2024-05-07:
Actually, I'm going to re-open this issue as I think its a deficiency in Bender.

Here is the scenario:
- we develop a CPU block that we want to use in different technology nodes.  As such, for various instantiated components such as clock buffers, particular memories, etc., we create a "generic" behavioural version of these components.  These are completely functional for simulation purposes, but some physical aspect of it is non-ideal.  For example, a behavioural memory array would be synthesized to a sea of flip-flops rather than a physical memory, but it will work.
- In ProjectA, we want to override these generic behavioural memory models and standard-cell models with vendor specific versions of these Verilog files with equivalent versions from (for example) TSMC
- In ProjectB, we want to instead do this with the equivalent from Global Foundries

In all cases, I don't want to change the CPU block, but simply remap these low-level standard functions to actual physical models instead.  To do this, I'd like to put the overrides into an upper level Bender.yml file that instantiates the CPU, and not a Bender.local file (which I don't think should be submitted to our Git database).

What are your thoughts on this?

micprog, 2024-05-08:
We do the same thing with our [tech_cells_generic](https://github.com/pulp-platform/tech_cells_generic) repository. For implementation, however, most technologies are not open-source, so we have internal technology-specific versions of these cells. They are overridden at the top level, however the methods vary (see https://github.com/pulp-platform/cheshire-ihp130-o for one example that is open-source).
I think checking in a `Bender.local` file into git is OK, and I would even recommend checking in the `Bender.lock` file that specifies the dependency git hash to ensure reproducibility. In my experience, the technology choice is generally done at the top level, so overriding at the implementation stage with the corresponding top-level is what I would recommend, not at any lower level. We generally also make use of targets to filter based on technology, which are also set at the top level.