pulp-platform/bender #185: syntax for target filtering dependencies?

epsilon537, 2024-10-15:
The README says: "Targets are flags that can be used to filter source files and dependencies."
I can't seem to figure out the syntax for target-based filtering of dependencies. I couldn't find any examples online either.
Thanks for any help.

micprog, 2024-10-17:
Targets can be specified in the `Bender.yml` with source file groups, as shown in the [Readme](https://github.com/pulp-platform/bender?tab=readme-ov-file#targets). To specify these targets for your scripts generation, add a `--target <YOUR_TARGET>` or `-t <YOUR_TARGET>` to your command. As an example, specifying the `test` target for a vsim script works as follows:
```
bender script vsim --target test
```
More info can be found in bender's help menu with `bender script --help`.

If you're looking for an example, check out [cheshire's](https://github.com/pulp-platform/cheshire) `Bender.yml` and [Makefile](https://github.com/pulp-platform/cheshire/blob/main/cheshire.mk#L142)

0xcdef, 2024-10-30:
it seems, now, you can list all the targets after '-t/--target' (one with each)  while do bender script

micprog, 2025-04-22:
Filtering dependencies with targets is currently not supported, and will be difficult to support to properly construct the dependency tree. You can exclude individual dependencies (and their exclusive dependents) in the script and sources commands with the `-e/--exclude <dep>` flag. Future features may add target filtering ability for dependencies similar to what you are proposing, although this will also still only apply with the sources and script commands that pass targets, not when updating the dependency tree.