pulp-platform/bender #191: Wrong syntax in Genus script

jagjordi, 2025-01-15:
I am generating a script for Genus, however the TCL syntax seems incorrect. See:
```
# This script was generated automatically by bender.
if [ info exists search_path ] {{
  set search_path_initial $search_path
}}
set ROOT = "/home/jordi/Projects/drra2/rtl"

set search_path $search_path_initial
```
there is an `=` in the `set ROOT` which is incorrect in TCL.

I don't know why this happens. I have checked the Genus template [here](https://github.com/pulp-platform/bender/blob/46ade8e6b97b8d27b601ed7348f541c399366c25/src/script_fmt/genus_tcl.tera#L7) and there is no `=`, so I don't know where it is coming from. Am I doing something wrong?

jagjordi, 2025-01-15:
On further inspection, it seems that this error only happens with the RHEL releases. I compiled with the latest git version and it works fine

micprog, 2025-04-22:
I believe this was fixed in #165, which should now be released in the `v0.28.2` binaries as well.

closed 2025-04-22 by micprog