pulp-platform/bender #198: Build fails on Ubuntu 24.04

hpretl, 2025-04-03:
Running our usual build process for IIC-OSIC-TOOLS now fails with:
```
0.178 [INFO] Building Bender
0.181 Cloning into 'bender'...
0.886 error: failed to parse lock file at: /tmp/bender/Cargo.lock
0.886
0.886 Caused by:
0.886   lock file version 4 requires `-Znext-lockfile-bump`
```

Here is our build script:
```
echo "[INFO] Building Bender"
git clone --depth=1 https://github.com/pulp-platform/bender.git
cd bender || exit 1
cargo build --release -j"$(nproc)"
```

@micprog @phsauter Please advise if we need to update something.

micprog, 2025-04-03:
This looks like you need to update your rust installation (`rustup update`)

hpretl, 2025-04-04:
I refactored our handling of Rust and Cargo, now it builds again.

closed 2025-04-04 by hpretl